SensorsIot/SuperPower #5: smart action on "Power Returns" ...

Andreas-Filsinger, 2020-09-12:
On "Power Loss" we have a Point where the "System" executes "poweroff" for a save shut down, every USV can do this.

Problems exists in the "Power returns" situation, nobody seems to care for this moment. The "System" should NOT start immediatly, only if there ist enough Power in the Accu to do a full "Cold Boot" PLUS a full "Shut down".

In many wrong implemented USVs the "System" is powered up with the return of the Power. But: If Power goes away (again) after say - 4 seconds - we have a incomplete Boot in the happy case, worst case is a Powerfail while running.
"SuperPower" should have selfmanaged Parameter "Systemstart if accu voltage > n V" where you can ensure that there is enough Power to do at least the "boot" and the "shut down".
Users should know this: The USV is OK if the System does not start if Power is plugged in - load the accu to say 10% is FIRST ORDER.

SensorsIot, 2020-09-14:
Good point. Maybe you include it in the discord features discussion. There is "where the music currently plays". I assume that, for the moment, not many people look at GitHub.

closed 2020-09-14 by SensorsIot

TD-er, 2020-09-14:
> [...] I assume that, for the moment, not many people look at GitHub.

Except those who are 'subscribed' to Github and always on Github as they don't use discord...

Just to add my 2 cents to the discussion, maybe it is possible to 'hard wire' a load test, by just triggering a resistor reflecting the true load by a fraction of a second to see if the voltage drops.
If so, then you're not measuring the charging voltage on a 'bad' battery.