SensorsIot/SuperPower #7: How about this for "SuperPower"?

vitalykorolev, 2020-09-17:
Adafruit just released a Universal USB / DC / Solar Lithium Ion/Polymer charger
Looks to me like it checking a lot of the right boxes, but I'm not an expert.
[https://www.adafruit.com/product/4755](https://www.adafruit.com/product/4755)
[image]

wassfila, 2020-09-19:
looks, good, this can go to the discord channel ready-made-boards, and I'll consider it for the competitive market analysis spreadsheet from the google drive.

rik94566, 2020-09-19:
here is one I found for RPi and thinking about giving a go.  I do not is it
on the list as yet, but might be good to evaluate as well.

https://www.robotshop.com/en/watchdog-battery-backup-raspberry-pi.html

Informaticore, 2020-11-15:
acknowledged closing

closed 2020-11-15 by Informaticore