SensorsIot/SuperPower #8: Consider using super capacitors instead of/additionaly to rechargable batteries.

amalic, 2020-09-22:
[Super capacitors](https://en.wikipedia.org/wiki/Supercapacitor) are a great alternative to batteries. They have very low discharge rate and are more efficient than batteries. 

As possible use case I could see solar powered enviromental sensor stations running on ESP32 in low power mode and doing one measurement every two hours. Once daily (around lunch time) transmitting the data through LoRa to a central server.

I'd be interested participating in such project.

wassfila, 2020-09-30:
you're welcome to participate, it starts with the discord channel, there you'd find links to the ongoing progress. github issues is not the path we defined for the requirements collection and analysis, but you can discuss them in the corresponding discord channel.

Informaticore, 2020-10-16:
This is a requirement request - we are already set with the current requirements but if you are willing to participate you can later use our repo to fork it and build a supercap variant out of it (to match the modularization concept) and we will be keen to have a look and add it as a variant if it fits.
Closing this ticket

closed 2020-10-16 by Informaticore