SensorsIot/SuperPower #44: Is the semiconductor crisis finally over?

adoconnection, 2024-11-18:
Hi all, it seems component shortages are doing to an end, so is there a way to make a new spin?

@SensorsIot @sethkaz @Tobi3566 @dr-stefan-krank @wassfila

SensorsIot, 2024-11-20:
The project team stopped existing long time ago :-(

XNinety9, 2024-11-20:
Sorry to read that :(
What's the project's status? Is its content useable @SensorsIot ?

SensorsIot, 2024-11-20:
The project has been stopped and the documentation is not up-to-date AFAIK

closed 2024-11-20 by SensorsIot

dr-stefan-krank, 2024-11-20:
Some of us are still here, but the project would have to be rolled up from the ground up. With the newer ESP32 models and their much improved deep sleep current and native USB connection, reviving this project isn't valuable anymore.
I'm here if something comes along that isn't covered by an already made board. Happy to be part of a project again.

sethkaz, 2024-11-20:
On the Raspberry Pi side, there are now a bunch of battery backup options available which meet many of the same features.  Admittedly, they aren't open-source, but they do meet that need.